thonny/thonny #2198: Backend issue

oliver2314, 2022-03-16:
Hi All,
I keep getting this issue with Thonny. I have uninstalled and re installed Thonny multiple times to receive the same issue. It seems to occur when i use my raspberry pi pico.

this is in the shell:Unable to connect to COM6: port not found

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

This occur when i press stop:

ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 136, in __init__
    self._prepare_after_soft_reboot(clean)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 169, in _prepare_after_soft_reboot
    self._sys_path = self._fetch_sys_path()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 442, in _fetch_sys_path
    return self._evaluate("__thonny_helper.sys.path")
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 611, in _evaluate
    raise ManagementError(script, out, err)
thonny.plugins.micropython.backend.ManagementError: Problem with a management command

SCRIPT:
__thonny_helper.print_mgmt_value(__thonny_helper.sys.path)

STDOUT:

STDERR:
Traceback (most recent call last):

  File "<stdin>", line 1, in <module>

AttributeError: 'module' object has no attribute 'path'

oliver2314, 2022-03-16:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 1961, in toolbar_handler
    handler(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 411, in cmd_run_current_script
    self.execute_current("Run")
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 370, in execute_current
    and not self._proxy.can_run_local_files()
AttributeError: 'NoneType' object has no attribute 'can_run_local_files'

 i also receive this

aivarannamaa, 2022-03-16:
Thank you for the report!

Have you tried selecting another port (Run => Select interpreter => MicroPython (Raspberry Pi Pico) => Port) ?

aivarannamaa, 2022-03-23:
Please reopen if you still have this problem

closed 2022-03-23 by aivarannamaa