thonny/thonny #2202: no moudle named 'tkinter'

wesley7891, 2022-03-18:
Hi, I am thonny user, thonny is a good ide and I use it to teach my 7 years old son python.

When I installed thonny in Ubuntu21.10, and opened the thonny, the terminal outputed this problem “no module named 'tkinter'”. 

I did some searching jobs in Google, then I fixed it.

```
sudo apt-get install python3-tk
```

I do program for a long time, this problem is not hard for me, but I think this problem may confuse the beginners.

If thonny's website add a install FQA page, it will help more people.

best regards

Wesley

aivarannamaa, 2022-03-18:
Thank you for the report!

You installed Thonny with pip, right?

wesley7891, 2022-03-18:
yes I use the pip to install the thonny , I think the pip didn't install the thonny's dependencies, but when I use apt-get to install the thonny. the Ubuntu's software repository didn't contain the thonny.