thonny/thonny #2203: Python 3.7.9 (bundled)

DBean0948, 2022-03-18:
Missing 'machine' module when I run the most basic blink scripts

from machine import Pin
from utime import sleep

led = Pin(25, Pin.OUT)

while True:
    led.on()
    sleep(0.5) # pause
    led.off()
    sleep(0.5)

hon 3.7.9 (bundled)
>>> %Run 03_01_blink.py
Traceback (most recent call last):
  File "C:\Users\dgies\Downloads\RB-Pico-List\03_01_blink.py", line 1, in <module>
    from machine import Pin
ModuleNotFoundError: No module named 'machine'

Please advise,,,Don Giesler beansbag@bellsouth.net

aivarannamaa, 2022-03-18:
Are you using CircuitPython or MicroPython on your Pico? CircuitPython doesn't have `machine` module, MicroPython does.

aivarannamaa, 2022-03-18:
Oh, from your stacktrace it looks like you are running the code with Python 3. You need to switch Thonny to either MicroPython or CircuitPython mode (the button in the lower-right corner of the main window)

aivarannamaa, 2022-03-23:
Please reopen if you still have this problem

closed 2022-03-23 by aivarannamaa

44yu5h, 2023-04-24:
Sorry to revive this issue but we cannot run MicroPython on RPi 3/4? 
Or there something I'm doing wrong? Sorry I'm just starting out with python. Thanks

[image]

aivarannamaa, 2023-04-25:
@44yu5h, on your screenshot I don't see Thonny. Have you posted the right screenshot?

44yu5h, 2023-04-25:
@aivarannamaa I'm sorry, but the response is same. here:
[image]

aivarannamaa, 2023-04-25:
@44yu5h, I don't see the lower part of Thonny's window, but I suspect you are trying to run MicroPython code in with CPython (Python 3) interpreter. You need to select a MicroPython device from the lower-right corner of the window.

44yu5h, 2023-04-25:
@aivarannamaa Yes you are right.
I just realised MicroPython depends on the hardware and I can't run it on Raspberry Pi 3/4 (only Pico) 
Correct me if I'm wrong. Also is there any way around I could be using the same library on RPi4? I haven't stumbled upon anything relevant so I'm assuming I cannot
Thanks!

[image]

aivarannamaa, 2023-04-25:
@44yu5h , you could try MicroPython Unix port: https://docs.micropython.org/en/latest/unix/quickref.html but I think you need to build it yourself for Rasperry. In Thonny the corresponding interpreter kind is "MicroPython (local)"

44yu5h, 2023-04-25:
@aivarannamaa Highly appriciate the efforts.
I'm not quite sure how to build it. I'll see how I can do so, I will let you know if I do
Thanks