thonny/thonny #2204: Thonny crashed

narektigranyan, 2022-03-18:
I have been using Thonny IDE, but it stopped working. Though I tried to fix the error with uninstalling and installing, but it is helpless. 

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny1\lib\site-packages\thonny\workbench.py", line 514, in _start_runner
    self._runner.start()
  File "C:\Program Files (x86)\Thonny1\lib\site-packages\thonny\running.py", line 122, in start
    self.restart_backend(False, True)
  File "C:\Program Files (x86)\Thonny1\lib\site-packages\thonny\running.py", line 612, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Program Files (x86)\Thonny1\lib\site-packages\thonny\plugins\cpython\__init__.py", line 284, in __init__
    super().__init__(clean, get_interpreter_for_subprocess(executable))
  File "C:\Program Files (x86)\Thonny1\lib\site-packages\thonny\plugins\cpython\__init__.py", line 40, in __init__
    super().__init__(clean, executable)
  File "C:\Program Files (x86)\Thonny1\lib\site-packages\thonny\running.py", line 877, in __init__
    self._start_background_process(clean=clean)
  File "C:\Program Files (x86)\Thonny1\lib\site-packages\thonny\running.py", line 944, in _start_background_process
    **extra_params
  File "C:\Program Files (x86)\Thonny1\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Program Files (x86)\Thonny1\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied

aivarannamaa, 2022-03-19:
Thank you for the report!

Could you please post your configuration.ini file? You can find it by selecting "Tools => Open Thonny data folder".

aivarannamaa, 2023-01-29:
Please reopen if you still have this problem with Thonny 4.0.2 and can provide more info.

closed 2023-01-29 by aivarannamaa