thonny/thonny #2207: Installation failed

speedy-10, 2022-03-21:
Hi, I tried the following:
`~/Downloads$ bash thonny-4.0.0b1.bash 

This script will download and install Thonny (thonny-4.0.0b1) for Linux (32 or 64-bit PC).
Press ENTER to continue or Ctrl+C to cancel.
Downloading https://github.com/thonny/thonny/releases/download/v4.0.0b1/thonny-4.0.0b1-x86_64.tar.gz
--2022-03-21 20:37:35--  https://github.com/thonny/thonny/releases/download/v4.0.0b1/thonny-4.0.0b1-x86_64.tar.gz
Auflösen des Hostnamens github.com (github.com) … 140.82.121.4
Verbindungsaufbau zu github.com (github.com)|140.82.121.4|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2022-03-21 20:37:36 FEHLER 404: Not Found.`

Regards

aivarannamaa, 2022-03-21:
Thank you for the report!

Linux tarball was unfortunately uploaded with wrong filename. Please try the same bash file again!

speedy-10, 2022-03-21:
Thanks, it works.
You do a great job  to consider all the reported issues!

closed 2022-03-22 by aivarannamaa