thonny/thonny #2208: Find & Replace doen't work

speedy-10, 2022-03-22:
Hi, if  I try to use 'crtl + F' there is no reaction of thonny, but the cursor moves one character to the right.

thonny 4.0.0b1

 A further observation:
- In the german menu Datei > "Move/rename" could be translated to "Verschieben/Umbenennen"
- in the menu  Extras > Optionen > Editor: there are missing some German translations, see picture.
->
[image]
 
Regards

aivarannamaa, 2022-03-23:
Thank you for the report! The problem is now fixed.

Regarding translations -- I don't write them myself. Please see https://github.com/thonny/thonny/blob/master/docs/translate.md#short-story-for-translators for the instructions.

closed 2022-03-23 by aivarannamaa

speedy-10, 2022-03-30:
ok, I registered myself here: https://poeditor.com/projects/po_edit?id_language=55&id=...
There are no strings left to translate (because beta?), so I'll wait for the next "full version" of thonny and support it then.