thonny/thonny #2209: Thonny 2.1 does not startup under Windows XP

ReyZRX, 2022-03-23:
Howdy. Yes, I know, XP is old...but it is on my laptop and only used on trips; at home use Win 10 Pro w/Py 3.8 (64bit) & 3.9 (32 bit).
Last Oct 2021, installed Thonny 2.1 under PIP after Python 3.4X installed. It ran w/no issues.
On my move to Indiana, it now does not startup.  Attached is PNG of err msg displayed.
Appreciate any suggestions.
Thanks, Rey
[image]

aivarannamaa, 2022-03-23:
Try the binary bundle, which has its own copies of suitable binaries: https://github.com/thonny/thonny/releases/download/v2.1.22/thonny-2.1.22.exe

ReyZRX, 2022-03-23:
Thanks. Will uninstall and try the binary bundle suggested and update you
on results.

Rey

>
> Message ID: ***@***.***>
>

deepeepee, 2023-01-02:
https://github.com/thonny/thonny/releases/download/v2.1.22/thonny-2.1.22.exe does not run in Windows XP!

aivarannamaa, 2023-01-02:
Unfortunately I don't have time for debugging issues specific to XP.

closed 2023-01-02 by aivarannamaa