thonny/thonny #2210: Addition of wrapping text and compiling certain sections under if, while, etc..

Sushi2k6, 2022-03-23:
Hello!
I am a big fan of thonny, I have been using it for the past 6 months. After comparing it with other IDEs, I found some more options which can be added to improve the user's experience.
 
1) The first one is this: [https://github.com/thonny/thonny/issues/1942](https://github.com/thonny/thonny/issues/1942)
    I thought that the issue has been solved but, it hasn't.
2) The ability to wrap text makes it so so much better, and I find it hard to check, if the code is beyond the screen.
3) Compiling lines of code of certain blocks (or) sections like under if, while, for, etc.
    This makes the code more organized, and simple
4) Showing the errors while typing the code [This is a big request and it's okay to ignore it too, just a suggestion]

Please fix these errors as soon as possible, especially number 1 and 2
Thanks :)

aivarannamaa, 2022-03-23:
Thank you for the suggestions!

These features are all nice to have, so I'll keep this issue in the TODO list, but I can't promise when will I get to them.

Sushi2k6, 2022-03-24:
Ok, but I would like for you to add this as soon as possible [https://github.com/thonny/thonny/issues/1942](https://github.com/thonny/thonny/issues/1942)