thonny/thonny #2211: தமிழ் (Tamil) Translation of Thonny UI

TamilNeram, 2022-03-24:
I would like to contribute to tamil translation for Thonny UI. I would like to know the steps for the same.

aivarannamaa, 2022-03-24:
Thank you for the offer! The steps are here: https://github.com/thonny/thonny/blob/master/docs/translate.md#short-story-for-translators

Sushi2k6, 2022-03-24:
Hi! I'm from Tamilnadu too. But Tamil translation is not needed. Because most of the coding is comfortable with English

TamilNeram, 2022-03-24:
Hi all,
        I have finished the Tamil translation. we never never know the
needs of all the Tamil users all around the world.

TamilNeram, 2022-03-24:
Hi Sushi2k6,
             உங்களை சந்தித்ததில் மகிழ்ச்சி. உங்கள் பெயர் என்ன? நான் இரண்டு
களஞ்சியங்களை உங்களுடன் பகிர்ந்து கொள்ள விரும்புகிறேன்.
1)https://github.com/TamilNeram/python - பைதான் நிரல்கள் தமிழில் -
importance of Python3 is Unicode support.
2)https://github.com/TamilNeram/TamilNeram.github.io which shows my github
home page https://tamilneram.github.io here.

அன்புடன்,                                          | Regards,
அனிஷ் பிரபு தி.                                   |  Anish Prabu T.

யாதும் ஊரே! யாவரும் கேளிர்!! <https://tamilneram.github.io/>
| World's
Citizen ! Everyone is my kin!! <https://tamilneram.github.io/>

First Indian CLA, CTA, + badges  <https://www.youracclaim.com/users/ta>

TamilNeram, 2022-03-25:
@aivarannamaa , 
                  Good morning, I have finished the tamil translation for Thonny. Is there a way i can check the functionality on the existing installtion of Thonny for Mac or windows 10. 

@Sushi2k6 ,
                shashidharan, I am Anish T.  Please find the list of portable apps that supports Tamil Language here. https://drive.google.com/drive/folders/1a0vSj603UvkMc37AOLSV5TQTSE2r46GC?usp=sharing.

aivarannamaa, 2022-03-25:
> Is there a way i can check the functionality on the existing installtion of Thonny for Mac

If you meant whether you can test your translations, then install https://github.com/thonny/thonny/releases/download/v4.0.0b1/thonny-4.0.0b1.pkg and follow the testing instructions at https://github.com/thonny/thonny/blob/master/docs/translate.md#short-story-for-translators (I just added information for finding Thonny's program directory)

TamilNeram, 2022-03-25:
Thanks @aivarannamaa  and @Sushi2k6 , It works fine. See the screenshot for details. highlighted the translated locations in green. rest or user file contents. 
[image]

TamilNeram, 2022-03-25:
Do I need to perform the File changes ? and commit for the languages.py file? for the "தமிழ்" entry in the dictionary.

[image]

Sushi2k6, 2022-03-25:
Thank you @TamilNeram. Really useful tool, sorry I couldn't reply soon cause I'm stuck up with exams.

@aivarannamaa Would like for you to fix this [https://github.com/thonny/thonny/issues/2214](https://github.com/thonny/thonny/issues/2214)

aivarannamaa, 2022-04-10:
Please test the translation in https://github.com/thonny/thonny/releases/tag/v4.0.0b2

closed 2022-04-10 by aivarannamaa