thonny/thonny #2212: Memory Error

philc52, 2022-03-24:
Hi. I was running a python file on a Pico for an extended period. Saving data to an SD card and also displaying on Shell. After some days, I received the following error dialog, which wouldn't close.
------------------------
Traceback (most recent call last):
  File "C:\Users\philj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1712, in event_generate
    handler(event)
  File "C:\Users\philj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 433, in _handle_program_output
    self._update_visible_io(None)
  File "C:\Users\philj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 512, in _update_visible_io
    self._apply_io_event(data, stream_name)
  File "C:\Users\philj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 648, in _apply_io_event
    self._applied_io_events.append((original_data, stream_name))
MemoryError
----------------------
I suspect that the error was caused by the continuous display (and memory) in the shell.
I will modify what is printed in the shell and see whether this allows my program to run longer.

aivarannamaa, 2022-03-24:
Thank you for the report!

I'll try to find the memory leak.

jarodriguez-itsoft, 2022-04-23:
I can confirm this problem.
Running on Windows 10, after some hours of debugging:

`Traceback (most recent call last):
  File "C:\Users\i7\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1712, in event_generate
    handler(event)
  File "C:\Users\i7\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1457, in _on_output
    self.append_text(msg["data"], stream_name)
  File "C:\Users\i7\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workdlg.py", line 243, in append_text
    setattr(self, stream_name, getattr(self, stream_name) + text)
MemoryError`