thonny/thonny #2213: Crashed and got this

Willow-Shade, 2022-03-24:
`INFO    thonny: Thonny version: 3.3.14
ERROR:root:Internal launch or mainloop error
Traceback (most recent call last):
  File "/home/ts/.local/lib/python3.8/site-packages/thonny/__init__.py", line 194, in launch
    from thonny import workbench
  File "/home/ts/.local/lib/python3.8/site-packages/thonny/workbench.py", line 15, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'
Traceback (most recent call last):
  File "/home/ts/.local/lib/python3.8/site-packages/thonny/__init__.py", line 194, in launch
    from thonny import workbench
  File "/home/ts/.local/lib/python3.8/site-packages/thonny/workbench.py", line 15, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ts/.local/bin/thonny", line 8, in <module>
    sys.exit(launch())
  File "/home/ts/.local/lib/python3.8/site-packages/thonny/__init__.py", line 214, in launch
    from thonny import ui_utils
  File "/home/ts/.local/lib/python3.8/site-packages/thonny/ui_utils.py", line 14, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'`

Any idea how to fix this?

Just started learning python and got crushed today.

aivarannamaa, 2022-03-24:
If you have a Debian based system, then you need to `sudo apt install python3-tk`. For other systems the package may be named `python3-tkinter`

Willow-Shade, 2022-03-24:
Ubuntu here. Sadly, none of them are working.

closed 2022-03-24 by Willow-Shade