thonny/thonny #2214: Error: Interpreter 'Äm鋸o' does not exist. Please check the configuration!

Sushi2k6, 2022-03-24:
[image]
This error pops everytime I try to run or debug or any other function

Sushi2k6, 2022-03-25:
I reinstalled the software as there was another problem. 
I am on 3.3.13
(It would be better if you ass the latest versions to the website)

Sushi2k6, 2022-03-25:
I reinstalled the software and then this happened.
I am on version 3.3.13
(It would be better if you add the latest versions to the website)

aivarannamaa, 2022-03-25:
what does it say in the lower-right corner of Thonny's main window?

Sushi2k6, 2022-03-25:
@aivarannamaa I would like for you to fix this issue as soon as possible, please. I can't use other debugs, because it is confusing.

TamilNeram, 2022-03-27:
Hi Shashidharan, 
              if you dont know the root cause of this issue. Try uninstalling and removing all files of Thonny from your system and then install again. 

For finding all files folder of Thonny,  use the "எல்லம்"(again Tamil Translation for this tool is from my team) from https://www.voidtools.com/.

Sushi2k6, 2022-03-27:
@TamilNeram 
I tried it, but again the error occurred.

aivarannamaa, 2022-03-28:
@Sushi2k6, please select "Tools => Open Thonny data folder" and send me your configuration.ini. It may contain clues for solving this problem.

Sushi2k6, 2022-03-29:
I even tried installing older versions, still has the same problem.

aivarannamaa, 2022-03-29:
@Sushi2k6, I requested access to your file in drive, but haven't received it

Sushi2k6, 2022-03-30:
@aivarannamaa Try now

aivarannamaa, 2022-04-01:
@Sushi2k6, thanks! Unfortunately I could not spot anything suspicious in the configuration file. Please try deleting this file (close Thonny first). After lauching Thonny again it should start with default settings. Does this make any difference?

aivarannamaa, 2022-04-01:
In this case you could try https://github.com/thonny/thonny/releases/tag/v4.0.0b1

Sushi2k6, 2022-04-02:
@aivarannamaa New issue, 
[image]
This occurred when I uninstalled the old one and installed the new version, and opened it.....

aivarannamaa, 2022-04-02:
@Sushi2k6, are you using Windows 7

Sushi2k6, 2022-04-03:
yes, and I know that this version cannot support, but I saw this and I thought that it will support Win 7
[image]

aivarannamaa, 2022-04-10:
Please try https://github.com/thonny/thonny/releases/download/v4.0.0b2/thonny-32bit-py38-4.0.0b2.exe

Sushi2k6, 2022-04-11:
but I have a 64bit os

aivarannamaa, 2022-04-11:
> but I have a 64bit os

32-bit Python works in 64-bit Windows as well.

aivarannamaa, 2022-04-11:
Did you try deleting the configuration.ini (while Thonny is closed) and then starting Thonny again?

aivarannamaa, 2022-04-13:
> what should I do now?

Try deleting the configuration.ini (while Thonny is closed) and then starting Thonny again (if you haven't tried it yet).

Sushi2k6, 2022-04-14:
I tried everything....

Sushi2k6, 2022-04-14:
This happened.....
[image]

aivarannamaa, 2022-04-14:
> I tried everything....

Just to be sure -- did you try deleting the configuration file while Thonny was closed? Please note that uninstalling does not delete this file.

> Is there anyway you can get the interpreter in a way I can install it externally?

If you have Windows 7, then you could install separate Python 3.8 (https://www.python.org/downloads/release/python-3810/) and then use "Run => Select interpreter" in Thonny to locate python.exe in this external installation.

Sushi2k6, 2022-04-15:
Yes, I tried that. I deleted the configuration.ini file while thonny was closed.

I don't get what you are trying to say, I should install python 3.8.10 ( I use that only ) and I should select it in the select interpreter?

aivarannamaa, 2022-04-16:
> I should install python 3.8.10 ( I use that only ) and I should select it in the select interpreter?

That's right.

Sushi2k6, 2022-04-21:
I tried it but a different error occurred. I meant to post this long back, I forgot

aivarannamaa, 2022-04-24:
At the moment I don't know how to proceed with this issue. If you have more information to add, then please re-open this issue or create a new one.

closed 2022-04-24 by aivarannamaa

Sushi2k6, 2022-04-25:
Ok, if I did have any other info, I will inform you.
Thanks

agorlach, 2024-01-23:
It is happen on my Win7 machine when I added api-ms-win-core-path-l1-1-0.dll  in hope to run the latest version of Thonny (4.1.4 x64 with Python 10).  I removed this version and tried to install x86 with Python 8. And then got this error. 

The error gone and Thonny start work when I removed api-ms-win-core-path-l1-1-0.dll from my machine.

kb4oam, 2024-12-16:
Removing all instances of 'api-ms-win-core-path-l1-1-0.dll' did the trick!
I've been fighting the 'interpretter problem' for weeks.
A big thanks!