thonny/thonny #2215: Management Error. Trying to install digi-xbee module into Raspberry Pi Pico

cranfo, 2022-03-25:
[image]

Tried with 2 different Raspberries, always the same error. Also tried cloning the module repository and uploading it directly into the Raspberry ('Upload to /' option).

cranfo, 2022-03-25:
[image]

"Could not write next block after having written 146432 bytes to /lib/digi/xbee/firmware.py
Make sure your device's filesystem has enough free space. (When overwriting a file, the old content may occupy space until the end of the operation.)" 

Must be a free space issue.

aivarannamaa, 2022-03-25:
It looks like https://pypi.org/project/digi-xbee is meant for CPython, not for MicroPython.

closed 2023-01-29 by aivarannamaa