thonny/thonny #2221: GUI locked after clipboard paste (Thonny 3.3.13, MacOS 12.2)

wintifrosch, 2022-03-31:
After pasting a clipboard content from a PDF to the shell console, UI becomes non responsible. But not frozen tough: Tooltipps still working, Menu commands still working (Save all, Quit, …), context menu appeared, but functions didn't work. Task manager doesn't report Tonny as «not responding». After 30 min UI was still frozen, had to kill the program.

Using Tonny 3.3.13 on MacOS 12.2 (21D49). Error may be reproduced without opened files, RaspberryPico board connected (in order to enable the shell frame) and pasting the string in the shell frame.
Steps to reproduce:
1. Start Thonny in macOS Finder
2. In Thonny, connect board, place cursor in Shell frame
3. Switch back to Finder, select [this PDF](https://dl.dropbox.com/s/1jdnxoq1ekv8sy7/CoffeeBreak_CheatSheet.pdf), open quick preview (=hit space bar), select page 5. 
5. select text in the third row «lambda x: x[0],» on the first table content line, copy this string to the clipboard
6. switch back to Thonny, Paste Clipboard

The PDF file was compiled from several one-page-PDFs using the MacOS Finder context menu function «Schnellaktionen > PDF erstellen» which appears if more than one PDF is selected (see 2nd screenshot below). The error does not appear if the text is taken from the [original one-page PDF](https://dl.dropbox.com/s/4yp3ndc9c86dlvc/CheatSheet-Python-5_-Functions-and-Tricks.pdf).

MacOS ActivityMonitor Task analyse report (German OS) [01](https://dl.dropbox.com/s/ri7p2o04xl8ig1w/01_Analyse_Text%E2%80%9Ecom.apple.appkit.xpc.openAndSavePanelService%20%28Thonny%29%E2%80%9C.txt), [02](https://dl.dropbox.com/s/ppi8os9oipzuq56/02_Anzahl_der_Analysen_%E2%80%9Ecom.apple.appkit.xpc.openAndSavePanelService%20%28Thonny%29%E2%80%9C.txt), [03](https://dl.dropbox.com/s/2out5rccop08453/03_Anzahl_der_Sekunden_%E2%80%9Ecom.apple.appkit.xpc.openAndSavePanelService%20%28Thonny%29%E2%80%9C.txt), [04](https://dl.dropbox.com/s/6sjpnhf9084s0ns/04_Prozent_des_uebergeordneten_Objekts%E2%80%9Ecom.apple.appkit.xpc.openAndSavePanelService%20%28Thonny%29%E2%80%9C.txt), [05](https://dl.dropbox.com/s/irbbyovm37ux6dg/05_Prozent_des_Threads%E2%80%9Ecom.apple.appkit.xpc.openAndSavePanelService%20%28Thonny%29%E2%80%9C.txt). 
Screenshots attached: (1) with selected Text (2) macOS Finder contact menu PDF compiling option and (3)–(5) Activity Monitor.
[image]
[image]
[image]
[image]
[image]

aivarannamaa, 2022-04-01:
Thank you for such a thorough report!

Before I dig in, could you please check, whether you can repeat this problem with https://github.com/thonny/thonny/releases/download/v4.0.0b1/thonny-4.0.0b1.pkg ?

aivarannamaa, 2022-04-01:
I'm afraid fixing these boxes and misplaced caret is out of my reach -- this should be taken to the developers of the GUI toolkit (Tk). Most likely they are aware of it already...

closed 2022-04-01 by aivarannamaa