thonny/thonny #2222: Open system shell does not work

git-aybo, 2022-03-31:
HI
Thank you for your great work with Thonny. 
I use Thonny 3.3.13 on Windows 10. When I choose the interpreter "Remote Python 3" it works fine with my Rasperry Pi in the Python environment. But when I choose Tools->Open system shell... it gives the error "command ssh not found". I would expect that a remote terminal would appear. In a normal windows terminal ssh works fine.
[image]
[image]

aivarannamaa, 2022-04-01:
Thank you for the report!

What do you get when you do `where ssh` in CMD?

git-aybo, 2022-04-01:
Hi
Please have a look to the 3rd image. You can see the output from where in CMD and Python

aivarannamaa, 2022-04-01:
Oh, sorry, I wasn't paying attention enough...

please try, whether giving the extensions helps:

```
>>> os.system("where ssh.exe")
>>> import shutil
>>> shutil.which("ssh.exe")
```

git-aybo, 2022-04-01:
[image]

aivarannamaa, 2022-04-01:
Please also investigate your PATHEXT evironment variable as described here: https://stackoverflow.com/questions/56399696/shutil-which-not-finding-programs-without-appending-file-extension

git-aybo, 2022-04-01:
This variable is set:
[image]
i also tried .exe

git-aybo, 2022-04-01:
The system shell from Thonny cannot reach the dircetory C:\Windows\System32\OpenSSH
[image]

git-aybo, 2022-04-01:
I think this is a window problem with 32/64 Bit (https://github.com/PowerShell/PowerShell/issues/11774 or https://www.reddit.com/r/csharp/comments/i06zdq/cant_access_cwindowssystem32opensshsshexe_from_c/):
Workaround: 
I use a own ssh and set the path variable in Thonny. After restart it works fine. 

Thank you for your support and have a nice day.

closed 2022-04-01 by aivarannamaa