thonny/thonny #2227: thonny-py5mode problem

lujo6911, 2022-04-04:
Python 3.10.4,   Processing 4.0b7,   py5 0.7.1a6,   thonny-py5mode 0.2.5a0

using imported_mode for small program, draw rectangle

The assistant reports:
Name "size" is not defined
Name "rect_mode" is not defined
... (for all Processing words)

Looks as if it run, but the sketch doesn't show

if I do not use imported mode with all the py5 prefixes it works.

tabreturn, 2022-04-05:
Thanks, @lujo6911 -- this isn't a Thonny issue but a plug-in one. I've logged an issue on thonny-py5mode (https://github.com/tabreturn/thonny-py5mode/issues/13), so you can close this one.

For anybody experiencing this problem, it's best to follow the [thonny-py5mode setup instructions](https://github.com/tabreturn/thonny-py5mode) and stick with [Thonny 3.3.7](https://github.com/thonny/thonny/releases/tag/v3.3.7) (the Python 3.9 'alt' versions) for now.

lujo6911, 2022-04-05:
> tabreturn/thonny-py5mode#13

Thanks,

Inserted code as in
https://tabreturn.github.io/code/python/thonny/2021/06/21/thonny_and_py5.html
,
but with the absolute path to site-packages to make it work:

        rel_filename = universal_relpath(script_path, working_directory)
        cmd_parts = ["%" + command_name, rel_filename] + args

######### py5 imported mode
        if args and args[0] == 'imported_mode':
            import site
            rs_file = '/py5_tools/tools/run_sketch.py'
            rs_path = "/usr/local/lib/python3.10/site-packages" + rs_file
            cmd_parts = ["%" + command_name, rs_path, rel_filename]
######### py5 imported mode

        exe_cmd_line = construct_cmd_line(cmd_parts,
[EDITOR_CONTENT_TOKEN]) + "\n"

Now I can use Thonny imported_mode, no import, no run_sketch, no py5.prefix.
Want to keep writing sketches simple, exploring Processing as a Scratch
follow-up track in a CoderDojo setting.

Thanks

aivarannamaa, 2022-04-10:
I'll test with Python 3.9 (for which jpype1 has published wheels), to see, if there is something to be done at Thonny's side.

aivarannamaa, 2022-04-23:
Leaved PR at py5mode's repo

closed 2022-04-23 by aivarannamaa