thonny/thonny #2229: Thonny won't open

EKE38bpm, 2022-04-04:
Hi Thonny peeps,
I'm an absolute novice, so please explain things as you would to a 6 year old!]

I've got a Chromebook and I'm attempting to install Thonny (again). 
I had it working and was using Thonny perfectly well, but had to do a 'Powerwash' so everything should go back to factory settings.

I've opened the Linux development environment, got Linux up and running and installed Thonny, but Thonny won't open. When I try to open it from the Launcher I get a very brief flash of the window which asks what language you want and what settings to use (I think), but not long enough to read or click anything.

From the Linux terminal:
~$ thonny
Traceback (most recent call last):
  File "/usr/bin/thonny", line 11, in <module>
    load_entry_point('thonny==3.2.7', 'gui_scripts', 'thonny')()
  File "/usr/lib/python3/dist-packages/thonny/__init__.py", line 149, in launch
    if not _check_welcome():
  File "/usr/lib/python3/dist-packages/thonny/__init__.py", line 122, in _check_welcome
    win = FirstRunWindow(mgr)
  File "/usr/lib/python3/dist-packages/thonny/first_run.py", line 12, in __init__
    super().__init__(className="Thonny")
  File "/usr/lib/python3.9/tkinter/__init__.py", line 2270, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0"

I have no idea what any of this means, but I just want Thonny back.
For what it's worth, I installed another Python IDE to see if it would work, and it does.

Help!!

Thanks
EKE

MumblingFumbler, 2022-04-06:
Same issue, same error.
Hardware is brand new Lenovo Chromebook
Linux install went fine.
I was able to install darktable no problems
I've tried installing both with
sudo apt install thonny
and
bash <(wget -O - https://thonny.org/installer-for-linux)
I've been able to successfully install Thonny on many (older) Chromebooks
doing a
export DISPLAY=unix$DISPLAY
as recommended here:
https://stackoverflow.com/questions/59006466/local-machine-tkinter-tclerror-couldnt-connect-to-display-0
got me a bit closer.
Thonny would run but the UI appeared as a single vertical line on the screen. I could drag to open a larger thonny window, but
it wouldn't stick across thonny runs.

Thanks for your help.

aivarannamaa, 2022-04-08:
Please see https://github.com/thonny/thonny/issues/2196 and https://bugs.python.org/issue47229

Can you confirm that IDLE is having the same problem on your computer?

Thanks for the stackoverflow link -- maybe I can use this for creating a work-around. If I only could reproduce this problem on my Chromebook...

MumblingFumbler, 2022-04-08:
Thanks for your reply.
Here's some more information.
The Linux distribution for my Chromebook is:
Debian Bullseye (v 11?)
Python version that comes installed in this distribution is
v 3.9.2
I installed VS Code with no apparent problems
I wrote a small python program
```
--------------------------------------------------
from turtle import *
print('Hello)
home()
forward(200)
--------------------------------------------------
```
Blew up on first line, apparently because tkinter package was not installed!
Installed tkinter using the following:
sudo apt-get install python3-tk
I was then able to run the above python code.
I don't understand how tkinter could not be included in the Bullseye distribution.
Shouldn't Python v 3.9.2 include tkinter, if it is necessary for the turtle package?

I've had to reinstall the Chromebook Linux container at least 5 times because
software installs have left Linux in an unknown state, and it is difficult to figure
out how to get it back to a known state.
So I'm treading cautiously in terms of installing Idle.
What method do you recommend for installing Idle?
When it doesn't work, what is the best way to get rid of it, and nothing else?
I'd be happy to provide any log files if they would be helpful.

Thanks for your help

aivarannamaa, 2022-04-09:
I could reproduce the problem after re-installing Crostini. Before re-install, my `DISPLAY` variable was set to `:0`, after re-install it was gone. Unfortunately simply setting it back did not help -- I got "could not connect to display :0"

For installing IDLE I used `sudo apt install idle3`. IDLE had same problem.

aivarannamaa, 2022-04-10:
I reported the issue at https://bugs.chromium.org/p/chromium/issues/detail?id=1315000

EKE38bpm, 2022-04-12:
Sorry for taking so long getting back to you. Kids seem to get in the way of my computer time!

I've installed Idle using sudo apt install idle3 but nothing seems to happen when I try to open it.
I guess I'll just have to try another Python IDE for now, but none of the ones I've tried in the past felt as good to me as Thonny did.

Thanks for looking into this. From what I've read (above) this doesn't look like a Thonny problem, but a Crostini problem.

Moosems, 2023-04-21:
@aivarannamaa According to [here](https://bugs.chromium.org/p/chromium/issues/detail?id=1315000#:~:text=This%20has%20been%20fixed%20and%20aiui%20the%20fix%20will%20be%20rolling%20out%20over%20the%20next%20few%20days%20(see%20bug%201314921).%20Sorry%20for%20the%20inconvenience.) the bug has been fixed, can you confirm?

aivarannamaa, 2023-04-21:
It was cornfirmed in #2196 but I forgot to update this issue. Just checked and Thonny 4.0.2 seems to run fine in my Chromebook.

closed 2023-04-21 by aivarannamaa