thonny/thonny #2234: Internal Error: invalid color name

Jgrothues, 2022-04-09:
I am a beginner, but I have used Thonny for some months now, without error. However, now, opening it always results in a fatal Internal Error. I have tried opening it in several different ways, but I have not been successful. Prior to the issue, the only things I changed on my device (Raspberry Pi, OS: Raspbian version 9) was installing some updates and changing the GUI appearance with lxappearance. 
Here is a copy of the error message:

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/__init__.py", line 65, in launch
    bench = workbench.Workbench(server_socket)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 151, in __init__
    self._init_menu()
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 317, in _init_menu
    self.get_menu("file", "File")
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1281, in get_menu
    menu = tk.Menu(self._menubar, **get_style_configuration("Menu"))
  File "/usr/lib/python3.5/tkinter/__init__.py", line 2720, in __init__
    Widget.__init__(self, master, 'menu', cnf, kw)
  File "/usr/lib/python3.5/tkinter/__init__.py", line 2151, in __init__
    (widgetName, self._w) + extra + self._options(cnf))
_tkinter.TclError: invalid color name `"#72cf"

```
I reinstalled multiple packages, reverted my changes on lxappearance, and rebooted the system, but the error persists. I am not sure how the error would be reproduced. I noticed it immediately after using lxappearance, although I my knowledge is not enough to confirm any correlation between it and the error.

Jgrothues, 2022-04-09:
Update:
I was messing about a bit with lxappearance and I got Thonny to work again. It seems evident that lxappearance affects Thonny in some way that causes a fatal error. Thonny only works if I use the setting of PiX.

aivarannamaa, 2022-04-09:
Thank you for the report!

I'll try to find ways for making Thonny survive bad colors coming from the system.