thonny/thonny #2235: I have problem with import requests

KritMath, 2022-04-09:
I try to write program to read Car Template but I got error. I already install requests. Could you please 
[image]

--------------------------------------------------------------------
import requests
 
url = "https://api.aiforthai.in.th/lpr-v2"
payload = {'crop': '1', 'rotate': '1'}
files = {'image':open(r"C:\Users\USER\Pictures\Car-Template.jpg", 'rb')}
 
headers = {
    'Apikey': "...........................................................",
    }
 
response = requests.post( url, files=files, data = payload, headers=headers)
 
print(response.json())
----------------------------------------------------------------------------------------------

Below that shown the error
------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\Users\USER\AppData\Roaming\Python\Python37\site-packages\requests\models.py", line 910, in json
    return complexjson.loads(self.text, **kwargs)
  File "C:\Program Files (x86)\Thonny\lib\json\__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "C:\Program Files (x86)\Thonny\lib\json\decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Program Files (x86)\Thonny\lib\json\decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\USER\Pictures\25650407 Krit - CarTemplate01.py", line 13, in <module>
    print(response.json())
  File "C:\Users\USER\AppData\Roaming\Python\Python37\site-packages\requests\models.py", line 917, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: [Errno Expecting value] : 0
----------------------------------------------------------------------------------------------

aivarannamaa, 2022-04-09:
It looks like your POST response is not a JSON

closed 2022-04-09 by aivarannamaa

KritMath, 2022-04-09:
How to solve it.

aivarannamaa, 2022-04-09:
I don't know. This issue tracker is for problems specific to Thonny

KritMath, 2022-04-09:
Thank you very much