thonny/thonny #2243: debugger stops/freezes with no error on Mac

pearlescentpanther, 2022-04-12:
Im running code in the debugger and usually after about 8 steps thrrough it will just stop and not let me continue stepping through - the step arrows turns grey. I can continue using the software afterward, but if I try again I get the same results. 
Im on OS Big Sur and I have only had this issue in Thonny v4.0.0b1

aivarannamaa, 2022-04-12:
Please also try the 2nd beta (https://github.com/thonny/thonny/releases/tag/v4.0.0b2) -- one debugger-related regression was fixed in b2.

pearlescentpanther, 2022-04-12:
i think that latest version has fixed the issue! perfect. looks like it had something to do with bringing up the pop up function window. thank you!

closed 2022-04-12 by pearlescentpanther

aivarannamaa, 2022-04-12:
Great! If you are happy with current state, then I'll close the issue.