thonny/thonny #2244: Please upgrade the `.py` installation

reza-ghazi, 2022-04-12:
`pip install thonnyapp` will raise the following warning:
```python
DEPRECATION: thonnyapp was installed using the legacy 'setup.py install' method, because a wheel could not be built for it. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
```

Please upgrade the installation process.

aivarannamaa, 2022-04-12:
I think I'll rather deprecate `thonnyapp` package, as it uses certain tricks for creating application shortcut, which doesn't fit into the preferred model of Python packages.

aivarannamaa, 2022-04-12:
In the meawhile just use `pip install thonny` and manually create the shortcut to `thonny` executable.

closed 2022-04-23 by aivarannamaa