thonny/thonny #2245: Problem with Debugging

AshDini, 2022-04-13:
Pressed Ctrl+F5
[image]
Pressed Ctrl + F5
[image]
Pressed Ctrl+F7
[image]
Pressed Ctrl + F6
[image]
Pressed Ctrl + F6
[image]
Pressed Ctrl + F6
[image]

aivarannamaa, 2022-04-13:
Thank you for the report! You are using 4.0.0b1, right? Please try https://github.com/thonny/thonny/releases/tag/v4.0.0b2

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem with version 4.0.2

closed 2023-01-28 by aivarannamaa