thonny/thonny #2247: Thonny locks up while editing

AshDini, 2022-04-13:
Thonny locks up while editing (using Thonny 4.0.0b2)
[image]

A little blue circle is circling, but it does not show on the screenshot.

aivarannamaa, 2022-04-18:
Thank you for the report!

After you force-close Thonny, can you repeat this situation? Can you give me example code and steps for reproducing this?

AshDini, 2022-04-20:
Hi Aivar,

It happened many times, but I could not find a common reason for it to
happen. Sorry I can not be more help.

Unfortunately, I had to switch back to the current (3.3.13) version

Regards,

Bob Stanton

chicory-ru, 2022-04-21:
You can repeat it by opening a large enough code and dragging the scroll up and down many times.

aivarannamaa, 2022-05-21:
I hope this issue was fixed in #2258, available since 4.0.0b3. Please reopen if you still get this error!

AshDini, 2022-10-11:
It just happened during editing the source code...

aivarannamaa, 2022-10-12:
@AshDini, with which version did it happen this time?

AshDini, 2022-10-15:
That was version 13. 14 seem to be ok.

aivarannamaa, 2022-12-28:
Assuming the issue is fixed in 4.0.1

closed 2022-12-28 by aivarannamaa

AshDini, 2022-12-28:
Yes, this issue have been fixed.