thonny/thonny #2249: no print output from remote via ssh

EriHau, 2022-04-16:
I am a beginner. Please forgive me if this problem is not a bug but a user error.
I run my python code on a headless raspberry pi and the thonny IDE runns on my Ubuntu-Laptop. Interpreter is "Remote Python 3 (SSH)". With the version 3.3.13 all works fine and I get all print outputs from my code and also error messages. 
Due to the file size error I switched to 4.0.0b2. Unfortunately I get no print output to the Shell window. The only message I get is ">>> %Run -c $EDITOR_CONTENT". But my code is running on my remote pi as I got messages over mqtt. An also when I put a syntax failure into my code there is no error message from the interpreter when trying to run the code. 
The windows which shows the remote files is also empty.   

**3.3.13:**
[image]

**4.0.0b2:**
[image]

aivarannamaa, 2022-04-18:
Thank you for the report!

Which OS are you using for Thonny? 

I can't reproduce this problem with simple `print`-s. Can you give me the simplest program that exposes this problem on your computer?

EriHau, 2022-04-18:
Thonny runs on **Ubuntu 20.04.4 LTS**
On my remote raspberry I have Raspbian 11 bullseye in headless mode. 

My simplest test-code:
```
import sys
print ("hello world")
with open('filename.txt', 'w') as f:
    print('hello filesystem', file=f)
```

You can see: no output
[image]

The file I can see at the remote:
[image]

But it's not only the print output, which is not shown. When I connect to the remote, there is also no line with the path- and version-information shown in the shell window. The shell is complete empty. No prompt, nothing. Just when I start the code, it looks like in the screenshot above.

aivarannamaa, 2022-04-18:
Thanks!

I found the bug. The fix will be in the next beta.

commit 9d8b785e (2022-04-18): Various SSH CPython fixes. Fixes #2249
closed 2022-04-18 by aivarannamaa