thonny/thonny #2251: microbit v2 with circuitpython thrown management error

puppet13th, 2022-04-19:
i'm still learning micropython using various boards like rp2040 and esp32 boards.Recently i got my microbit v2 board and installed micropython firmware from microbit website.That micropython firmware was so limited in functionality,so i seek another alternative and i found circuit python.when microbit uses circuitpython,thonny got management error:
```
Adafruit CircuitPython 7.2.5 on 2022-04-06; micro:bit v2 with nRF52833
>>> 
NB! Thonny could not execute 'get_dirs_children_info'.

SCRIPT:
__thonny_helper.print_mgmt_value({name : __thonny_helper.os.size(name) for name in __thonny_helper.os.listdir()})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in <dictcomp>
AttributeError: 'module' object has no attribute 'size'

You may need to reconnect, Stop/Restart or hard-reset your device.
```
Look like thonny still works with micropython backend, i have already selected circuitpython and correct com port.
i installed another circuitpython and it works well with thonny.
```
Adafruit CircuitPython 7.2.5 on 2022-04-06; Waveshare RP2040-Zero with rp2040
>>> 
Backend terminated or disconnected. Use 'Stop/Restart' to restart.

──────────────────────────────────────────────────────────────────────────────────
Adafruit CircuitPython 7.2.5 on 2022-04-06; Waveshare RP2040-Zero with rp2040
>>> 
```
Mu editor works well with microbit v2 + circuitpython installed.i'm hoping that i could use thonny with microbit v2 + circuitpython.

aivarannamaa, 2022-04-19:
Thank you for the report!

I suspect that Thonny only considers "micro:bit" in "Adafruit CircuitPython 7.2.5 on 2022-04-06; micro:bit v2 with nRF52833" and thinks it needs to adapt for the MicroPython API for micro:bit.

I'll try to fix this before the next beta of Thonny 4.0

puppet13th, 2022-04-20:
like i thought,hehe, i'm looking forward for the fixes. thank you very for your hard work.

aivarannamaa, 2022-04-23:
Fix will be in the next beta.

closed 2022-04-23 by aivarannamaa

puppet13th, 2022-05-22:
confirmed fixed using [Thonny 4.0.0b3](https://github.com/thonny/thonny/releases/tag/v4.0.0b3)
```
Adafruit CircuitPython 7.2.5 on 2022-04-06; micro:bit v2 with nRF52833
Adafruit CircuitPython 7.2.5 on 2022-04-06; micro:bit v2 with nRF52833

>>> help('modules')
__future__        audiopwmio        math              rtc
__main__          binascii          microcontroller   storage
_bleio            board             micropython       struct
aesio             builtins          onewireio         supervisor
analogio          busio             os                synthio
array             collections       pulseio           sys
atexit            digitalio         pwmio             time
audiobusio        errno             rainbowio         touchio
audiocore         gc                random            traceback
audiomixer        getpass           rotaryio          watchdog
Plus any modules on the filesystem
>>> 
```