thonny/thonny #2253: pybabel fails compiling the localization [RPM PACKAGING]

abrarwali, 2022-04-20:
`pybabel compile`  compiles the localization, and it fails.
TRACEBACK:
+ pybabel compile -d thonny/locale/ -D thonny
compiling catalog thonny/locale/ar_AR/LC_MESSAGES/thonny.po to
thonny/locale/ar_AR/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/cs_CZ/LC_MESSAGES/thonny.po to
thonny/locale/cs_CZ/LC_MESSAGES/thonny.mo
error: thonny/locale/de_DE/LC_MESSAGES/thonny.po:902: placeholders are
incompatible
error: thonny/locale/de_DE/LC_MESSAGES/thonny.po:1388: placeholders are
incompatible
compiling catalog thonny/locale/de_DE/LC_MESSAGES/thonny.po to
thonny/locale/de_DE/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/el_GR/LC_MESSAGES/thonny.po to
thonny/locale/el_GR/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/en_US/LC_MESSAGES/thonny.po to
thonny/locale/en_US/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/es_ES/LC_MESSAGES/thonny.po to
thonny/locale/es_ES/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/et_EE/LC_MESSAGES/thonny.po to
thonny/locale/et_EE/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/fa_IR/LC_MESSAGES/thonny.po to
thonny/locale/fa_IR/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/fi_FI/LC_MESSAGES/thonny.po to
thonny/locale/fi_FI/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/fr_FR/LC_MESSAGES/thonny.po to
thonny/locale/fr_FR/LC_MESSAGES/thonny.mo
error: thonny/locale/hu_HU/LC_MESSAGES/thonny.po:1388: incompatible
format for placeholder 1: 's' and 'i' are not compatible
compiling catalog thonny/locale/hu_HU/LC_MESSAGES/thonny.po to
thonny/locale/hu_HU/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/hy_AM/LC_MESSAGES/thonny.po to
thonny/locale/hy_AM/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/it_IT/LC_MESSAGES/thonny.po to
thonny/locale/it_IT/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/ja_JP/LC_MESSAGES/thonny.po to
thonny/locale/ja_JP/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/ko_KR/LC_MESSAGES/thonny.po to
thonny/locale/ko_KR/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/lt_LT/LC_MESSAGES/thonny.po to
thonny/locale/lt_LT/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/nb_NO/LC_MESSAGES/thonny.po to
thonny/locale/nb_NO/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/nl_NL/LC_MESSAGES/thonny.po to
thonny/locale/nl_NL/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/nn_NO/LC_MESSAGES/thonny.po to
thonny/locale/nn_NO/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/pl_PL/LC_MESSAGES/thonny.po to
thonny/locale/pl_PL/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/pt_BR/LC_MESSAGES/thonny.po to
thonny/locale/pt_BR/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/pt_PT/LC_MESSAGES/thonny.po to
thonny/locale/pt_PT/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/ro_RO/LC_MESSAGES/thonny.po to
thonny/locale/ro_RO/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/ru_RU/LC_MESSAGES/thonny.po to
thonny/locale/ru_RU/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/sk_SK/LC_MESSAGES/thonny.po to
thonny/locale/sk_SK/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/sq_AL/LC_MESSAGES/thonny.po to
thonny/locale/sq_AL/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/sv_SE/LC_MESSAGES/thonny.po to
thonny/locale/sv_SE/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/ta_IN/LC_MESSAGES/thonny.po to
thonny/locale/ta_IN/LC_MESSAGES/thonny.mo
error: thonny/locale/th_TH/LC_MESSAGES/thonny.po:1854: placeholders are
incompatible
compiling catalog thonny/locale/th_TH/LC_MESSAGES/thonny.po to
thonny/locale/th_TH/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/tr_TR/LC_MESSAGES/thonny.po to
thonny/locale/tr_TR/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/uk_UA/LC_MESSAGES/thonny.po to
thonny/locale/uk_UA/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/zh_CN/LC_MESSAGES/thonny.po to
thonny/locale/zh_CN/LC_MESSAGES/thonny.mo
compiling catalog thonny/locale/zh_TW/LC_MESSAGES/thonny.po to
thonny/locale/zh_TW/LC_MESSAGES/thonny.mo
4 errors encountered.
error: Bad exit status from /var/tmp/rpm-tmp.yuMwcf (%build)
Bad exit status from /var/tmp/rpm-tmp.yuMwcf (%build)

befeleme, 2022-04-21:
Hi, I edited the respective lines in POEditor (it was mostly about overlooked placeholders), so just importing the fresh strings for DE, HU and TH should fix the compilation problem.

aivarannamaa, 2022-04-21:
@befeleme, thanks for this observation and the fix!

@abrarwali, are you interested in the head of the master branch or specific version / tag?

abrarwali, 2022-04-21:
@aivarannamaa I'm interested in v 4.0.0b2

abrarwali, 2022-04-24:
@aivarannamaa Version 4

closed 2022-08-14 by aivarannamaa