thonny/thonny #2258: ‘pythonw’ interpreter cannot re-run tk program

TooLazy0x00, 2022-04-25:
‘pythonw’ interpreter cannot re-run tk program

Nice application, especially autocompletion.

In the process of using 4.0.0.b2, I found that if the UI zoom factor was adjusted to 2, the program would freeze and exit.

Also, when using the 'pythonw.exe' interpreter to run the tkinter program, pressing F5 will not re-run the program, but will give a reminder in the shell
--'Could not interrupt current process. Please wait, try again or select Stop/Restart!'--.

At this time, you can only press 'ctrl f2' to restart.

Also I would like to ask, is there any way to resize the shell window from the bottom of the screen to the right of the screen?

aivarannamaa, 2022-04-25:
Thank you for the report!

> In the process of using 4.0.0.b2, I found that if the UI zoom factor was adjusted to 2, the program would freeze and exit.

You mean "UI scaling factor", right? Did it freeze right after pressing "OK" or after restarting Thonny? Did it run correctly next time you started Thonny?

> Also, when using the 'pythonw.exe' interpreter to run the tkinter program, pressing F5 will not re-run the program, but will give a reminder in the shell
> --'Could not interrupt current process. Please wait, try again or select Stop/Restart!'--.

This is expected as pythonw.exe doesn't have console for sending the interrupt signal.

> when I change 'python.exe' interpreter
> --PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_frame_info' (AttributeError: module 'sys' has no attribute 'last_traceback').

Which Python version did you select? Same as bundled with Thonny? Did you install the bundle with 32-bit Python 3.8 or 64-bit Python 3.10 (the default) ?

> Also I would like to ask, is there any way to resize the shell window from the bottom of the screen to the right of the screen?

Please see https://github.com/thonny/thonny/wiki/Custom-layout

TooLazy0x00, 2022-04-26:
Thank  you, the shell has successfully adjusted to the right side of the screen.

I installed thonny via pip，python 3.8.10 64bit

Unfortunately, the app exit caused by UI scaling did not happen again.

I tested it again and found that the stuck exit occurs in the following situations.

Open 2 py files, for example, I open a py file written by myself, then press View-Files, switch directories, and open python/lib/site-packages/thonny/shell.py, which can be opened normally, but when the mouse wheel looks down file about line200, thonny will exit directly，this happens all the time.

Maybe adjusting the UI coincides with this situation

aivarannamaa, 2022-04-28:
Unfortunately I can't reproduce this.

While you're scrolling in shell.py, do you (occasionally) see the horizontal scroll-bar of the editor? Can you reproduce the problem when you make other panes narrow enough (or Thonny's main window wide enough) such that the horizontal scroll-bar doesn't appear?

TooLazy0x00, 2022-04-29:
When there was an error, the computer didn't show the traceback, it just exited.

I changed a computer and did some tests. The test conditions are as follows:
Start thonny through cmd, open shell.py
shell.py-N
view-files-W
view-shell-S
view-outline-E
When I swipe down the code page with the mouse wheel, it comes up with a traceback like this:
Line 14 "_tkinter.TclError: named font font17 does not already exist" will change like this -->font16

Traceback (most recent call last):
  File "D:\python\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "c:\users\administrator\python\lib\site-packages\thonny\tktextext.py", line 1129, in _horizontal_scrollbar_update
    self.update_margin_line()
  File "c:\users\administrator\python\lib\site-packages\thonny\tktextext.py", line 1075, in update_margin_line
    get_text_font(self.text).measure((margin_line_visible_col - 1) * "M")
  File "c:\users\administrator\python\lib\site-packages\thonny\tktextext.py", line 1160, in get_text_font
    return tkfont.nametofont(font)
  File "D:\python\lib\tkinter\font.py", line 22, in nametofont
    return Font(name=name, exists=True)
  File "D:\python\lib\tkinter\font.py", line 86, in __init__
    raise tkinter._tkinter.TclError(
_tkinter.TclError: named font font17 does not already exist

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\python\lib\tkinter\__init__.py", line 1462, in _windowingsystem
    return self._root()._windowingsystem_cached
  File "D:\python\lib\tkinter\__init__.py", line 2354, in __getattr__
    return getattr(self.tk, attr)
AttributeError: '_tkinter.tkapp' object has no attribute '_windowingsystem_cached'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\administrator\python\lib\site-packages\thonny\__init__.py", line 250, in launch
    bench.mainloop()
  File "D:\python\lib\tkinter\__init__.py", line 1429, in mainloop
    self.tk.mainloop(n)
  File "D:\python\lib\tkinter\__init__.py", line 1896, in __call__
    self.widget._report_exception()
  File "D:\python\lib\tkinter\__init__.py", line 1612, in _report_exception
    root.report_callback_exception(exc, val, tb)
  File "c:\users\administrator\python\lib\site-packages\thonny\workbench.py", line 2388, in _on_tk_exception
    self.report_exception()
  File "c:\users\administrator\python\lib\site-packages\thonny\workbench.py", line 2400, in report_exception
    dlg = ui_utils.LongTextDialog(title, msg, parent=self)
  File "c:\users\administrator\python\lib\site-packages\thonny\ui_utils.py", line 1795, in __init__
    super().__init__(master=parent)
  File "c:\users\administrator\python\lib\site-packages\thonny\ui_utils.py", line 45, in __init__
    if self._windowingsystem == "aqua":
  File "D:\python\lib\tkinter\__init__.py", line 1465, in _windowingsystem
    self.tk.call('tk', 'windowingsystem')
_tkinter.TclError: too many nested evaluations (infinite loop?)

TooLazy0x00, 2022-04-29:
As you said, when I set the window to look like this:
shell.py - N
view -shell -S
There is no horizontal scroll bar in the window, the program has been running normally, and has not been exited again.

aivarannamaa, 2022-04-29:
Thanks for the stacktrace! It mentions the margin line indicating the recommended line length, and when I turned it on, I could reproduce the freeze and exit.

TooLazy0x00, 2022-04-29:
The situation is almost as follows：

When the font of the code window is enlarged to a certain value, the horizontal scroll bar does not appear at this time.

Left-click on the vertical scroll bar and scroll down line by line, will encounter a situation where the code exceeds the window. At this time, the horizontal scroll bar flashes rapidly. 

If continue to operate the vertical scroll bar, it will be fine. If operate other buttons, there will be no response & quit

aivarannamaa, 2022-04-30:
I implemented a solution for the wobbling scrollbar (not hiding it anymore after it becomes visible). 

I could not reproduce the stacktrace you gave me, but I guarded this part of code by falling back to not updating the margin line in case of similar exception.

closed 2022-04-30 by aivarannamaa