thonny/thonny #2259: Not Recognizing Walrus Operator :=

ahowes83, 2022-04-25:
[image]
Thonny is throwing an error when I try to use a := here. Same code works in VS and comes from the new CS50p video from Harvard.

`import re

url = input("URL: ").strip()

if matches := re.search("^(?:https?://)?(?:www\.)?twitter\.com/([a-z0-9_]+)", url, re.IGNORECASE):
    print(f"Username:", `matches.group(1))`

aivarannamaa, 2022-04-26:
Thonny 3.3 is bundled with Python 3.7, which doesn't recognize the walrus operator (introduced in Python 3.8).

The solution is install separate Python 3.8 (or newer) and make Thonny use it (Run => Select interpreter) or use the beta of Thonny 4.0 (https://github.com/thonny/thonny/releases)

ahowes83, 2022-04-26:
Ah, thanks very much -- I'm not too surprised it was my mistake and not a real issue -- Andy

closed 2022-04-26 by ahowes83