thonny/thonny #2260: File browser handling of links

dbuggz, 2022-04-27:
So I see that under Windows, Thonny uses APDATA to open the default C:\Users\username and since I have a few devices I use Thonny with (MicroBit, pico, badger) I'd like to have a quick way of linking to my DATA drive (D:) and the directories I keep these different devices/projects in. I have tried creating a Thonny directory (C:\Users\username\Thony) with links to D:\Documents\Pimoroni\Badger etc. but when you click on that it opens Windows Explorer at that directory instead of following the link.

aivarannamaa, 2022-04-28:
Thank you for the report!

For replicating it -- can you tell me precisely, how did you create the links? With `mklink`? (I've never used symlinks in Windows myself)

aivarannamaa, 2023-01-28:
Please reopen if you're still interested in this problem and can add more information.

closed 2023-01-28 by aivarannamaa

dbuggz, 2023-01-28:
FYI this took a few attempts at WinDOZE stuff. Finally the command I used was :
_C:\Users\davew\Thonny>mklink /j PicoDir "D:\Documents\Raspberry Pi\Pico"
Junction created for PicoDir <<===>> D:\Documents\Raspberry Pi\Pico_