thonny/thonny #2262: Problem with backend?

mbleight, 2022-04-27:
Traceback (most recent call last):
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1449, in event_generate
    handler(event)
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 223, in _handle_program_output
    self._insert_text_directly(msg.data[1:], ("io", msg.stream_name))
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 447, in _insert_text_directly
    _insert(txt, tags)
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 428, in _insert
    self.direct_insert("output_insert", txt, tags)
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\ui_utils.py", line 677, in direct_insert
    self, index, chars, tags=tags, **kw
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 721, in direct_insert
    super().direct_insert(index, chars, tags, **kw)
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 167, in direct_insert
    self._original_insert(index, chars, tags, **kw)
  File "C:\Users\leigh\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 43, in original_function
    self.tk.call((self._original_widget_name, operation) + args)
_tkinter.TclError: character U+1f60d is above the range (U+0000-U+FFFF) allowed by Tcl

aivarannamaa, 2022-04-28:
Do you use emoji characters in your code? Which Thonny version are you using? You could try the betas of Thonny 4.0, which have newer Tk for better Unicode support: https://github.com/thonny/thonny/releases

mbleight, 2022-04-28:
> 

So I am trying to scrape twitter data for sentimental analysis and use regex to simplify the expressions but yes i do use emojis. Not sure why but using idle solved my problem so i think im good. Thank you though

closed 2022-04-28 by mbleight