thonny/thonny #2263: Intune validation string not reflecting version

TimButterworth, 2022-04-29:
Hi There,

I'd like to use Intune to manage Thonny on our pupil 1-to-1 devices, but there seems no obvious way for me to validate the installation according to software version. The File properties on the .exe are 1.0.0.1 rather than the version number, for example, so we won't be able to use that for the supersedence function for updates.

Any advice would be welcome,

Thanks!

Tim

aivarannamaa, 2022-04-30:
thonny.exe is just the launcher, which does not change between versions, and I'd rather not change and re-sign it just for updating the version string.

Could you take the version string from `lib\site-packages\thonny\VERSION` ?

TimButterworth, 2022-05-03:
> CAUTION: This email originated from outside ESMS. Do not click links or open attachments unless you recognise the sender and know the content is safe.

Thanks for the reply - that should do the trick!

Kind regards,

Tim

ESMS
A unique family of independent schools in Edinburgh for children aged 3-18
www.esms.org.uk<http://www.esms.org.uk/>

[ESMS Facebook]<https://www.facebook.com/esmsedinburgh> [ESMS Twitter] <https://twitter.com/esmsedinburgh>  [ESMS Instagram] <https://www.instagram.com/esmsedinburgh>  [ESMS Vimeo] <https://vimeo.com/esmsedinburgh>  [ESMS Flickr] <https://www.flickr.com/photos/esmsedinburgh>

This e-mail and any attachments are private and confidential and may be protected by copyright. If you are not the intended recipient please inform the sender and delete this message. Information in this message which does not relate to the business of Erskine Stewart's Melville Schools (ESMS) is not authorised by ESMS which accept no liability for such content.

This e-mail and any attachments have been checked for virus content. However it is the responsibility of the recipient to satisfy themselves in this regard. ESMS are not responsible for any consequent loss or damage arising from any virus or from any disabling device contained in the e-mail. ESMS reserve the right to monitor all email communication through their network.

ESMS are independent schools operating under the aegis of the Merchant Company Education Board, Reg. Charity No. SC 009747, whose offices are at 22 Hanover Street, Edinburgh, EH2 2EP, Scotland.

closed 2023-01-28 by aivarannamaa