thonny/thonny #2265: Some minor deviations to standard editors

street-grease-coder, 2022-04-30:
There are some annoyances with Thonny that, IMHO, keeps it from being absolutely amazing.

1) CTRL+SHIFT+Arrow keys do not follow default behavior of notpad. E.g. in this example:
`def test(): can I write[1]?| ` with cursor at `|` and `[CTRL+SHIFT+ARROW LEFT]` unintuitively marks `1]?` (default windows notepad marks `[1]?`)
`def test():| can I write[1]?` with cursor at `|` and `[CTRL+SHIFT+ARROW LEFT]` unintuitively marks `test()` (default windows notepad marks `()?`)

2) if you open up the text search window and switch the tab, the search context still remains in the original tab, when it intuitively should focus on the currently open tab

3) sometimes Thonny crashes when longer commands are interrupted and with several mutliline commands, resulting in a never-ending error 'beep', that can only be resolved by killing the process. Will look for error log next time this happens.

aivarannamaa, 2022-05-04:
Thank you for the suggestions! I'll fix the search thing before final 4.0.

Arrow key behavior is more tricky, as it's currently implemented inside the GUI toolkit.

In 3rd point, did you mean submitting multiline commands in the Shell pane?

aivarannamaa, 2023-01-29:
Search is fixed in 4.0.2.

Caret movement is hard to fix, maybe sometime in the future...

Please open another issue for 3rd point if you experience it again and can provide the logs.

closed 2023-01-29 by aivarannamaa