thonny/thonny #2266: Kivy Windows Don't Close

Ardemus, 2022-05-01:
Hello, here's my Thonny environment:

Thonny 3.3.13 (fresh install using the bash scrip from the project website)
Pop_OS! 21.10
Python 3.7.9
Tk 8.6.10
Kivy 2.1.0 (installed through the Thonny system shell using the [full] dependencies option as described on the [kivy website](https://kivy.org/doc/stable/gettingstarted/installation.html) (I did not create a virtual environment).
Nothing else

Running through the Kivy [Hello World](https://kivy.org/doc/stable/guide/basic.html) and [Pong](https://kivy.org/doc/stable/tutorials/pong.html) Tutorials I found that the close button (in the linux title bar) will stop the app, but not clear the window. Clicking the stop button in Thonny does clear the window.

It's an annoyance, but is suggests a problem somewhere that may crop up in other ways.

aivarannamaa, 2022-05-01:
Thank you for the report!

> will stop the app, but not clear the window

What do you mean by clearing the window? Which window? Does your Kivy window close?

Note that Thonny does not end the Python process when the main script ends (just as if you had run the script with `python -i`). This may make a difference.

If you don't like Thonny's approach, then you may prefer "Run => Run current script in terminal" instead of "Run => Run current script". See also the options at "Tools => Options => Terminal".

Ardemus, 2022-05-02:
Yes, after using Thonny for a few more days it seems like it's operating as intended. Thonny provides some debug/linting information for review after the code exits. Presumably it needs to keep things alive to do that. When I press Stop it finishes shutting things down as expected.

closed 2022-05-02 by Ardemus

aivarannamaa, 2022-05-03:
I experimented with Kivy a bit and found that you can get rid of Kivy window even in Thonny when you add `Window.close()` to the end of your script:

```python
import kivy
kivy.require('2.1.0') # replace with your current kivy version !

from kivy.app import App
from kivy.uix.label import Label
from kivy.core.window import Window

class MyApp(App):

    def build(self):
        return Label(text='Hello world')

if __name__ == '__main__':
    MyApp().run()
    Window.close()
```