thonny/thonny #2267: Installing locally on a mac

isaacl, 2022-05-01:
Hello!  I am wondering if there is a mac install method which does not modify the machine's global python structure. 

I use pyenv and am nervous that a thonny install will mess up my paths.  On Windows there is the option for a user install, which won't affect global python installs, etc.  I do not see this option on mac.

The graphics from [SeparateInstall](https://github.com/thonny/thonny/wiki/SeparateInstall) seem more primitive than the official mac app.

aivarannamaa, 2022-05-01:
No worries -- Thonny installer neither depends on nor messes with system Python. It comes with its own Python framework inside the app bundle.

isaacl, 2022-05-02:
Why does the thonny installer need root privileges?

aivarannamaa, 2022-05-02:
I believe it asks for admin privileges because it has option for installing for all users. I don't know why it doesn't ask it (or skip asking it) after selecting the target location. I haven't found documentation about such details. Unfortunately I can't use simple drag-and-drop installation method, because certain scripts must be modified according to installation location after installation.

isaacl, 2022-05-02:
I'd recommend a note page in the installer pane before you ask, if possible!  It is a scary prompt for an app that "feels" like it should be draggable!

isaacl, 2022-05-02:
> I don't know why it doesn't ask it (or skip asking it) after selecting the target location. 

It asks immediately, any options.

aivarannamaa, 2022-05-21:
I added this explanation to the readme file of the installer

closed 2022-05-21 by aivarannamaa