thonny/thonny #2269: Debug windows are too wide

isaacl, 2022-05-02:
4b2 on Windows 10.

When I debug, the function pop ups are much wider than the v3 screenshots on thonny.org.

[image]

aivarannamaa, 2022-05-03:
The width of these windows depend on the width of the code editor.

isaacl, 2022-05-04:
It looks weird for a popup window imo

aivarannamaa, 2022-05-04:
> It looks weird for a popup window imo

It does, but following the editor's width is the easiest heuristics for predicting the required width in the stack windows (can't simply check the longest line in the main script, as the functions may be imported from other files)

isaacl, 2022-05-10:
>  following the editor's width is the easiest heuristics for predicting the required width

Hmm, not sure I follow that.  The debugger might jump into wide code that isn't part of the editor.  Or wide code might be part of the editor but passed over in the debugger.

What about a max-width for the pop-up windows, say 140 or 160 characters?

aivarannamaa, 2022-05-13:
For the sake of simplicity (and also for saving time for more pressing issues) I'd prefer to keep it as is (for now). I guess most users who don't mind too wide editors also don't mind too wide frame windows. You still have the option to make your editor narrower, either by reducing the overall width of Thonny's main window or by increasing the width of the left and right panes.

closed 2022-05-13 by aivarannamaa