thonny/thonny #2272: Pi Pico firmware update

Frank-99, 2022-05-05:
Every time I connect the RP2040-Zero board and press Stop/Restart, it wants to reinstall the firmware, which is ok I guess.
Problem is that I live in the italian countryside, so not a good internet connection at all sometimes. Is there really no way to download the firmware on my PC and have the option to install that local firmware instead of fetching it online every time?
on bad-internet days I simply can't work on my controller boards because the request goes straight to timeout. is there another way to do it offline and i just missed it, or am i screwed basically in these occasions?
Thank you in advance
- Diego

aivarannamaa, 2022-05-05:
I assume you meant RP2040-Pico, not Zero.

How do you connect it? Thonny is supposed to offer firmware reinstall only if the board is connected while holding the BOOTSEL button. If you already have MicroPython installed on it then you should plug it without holding BOOTSEL.

Frank-99, 2022-05-05:
> I assume you meant RP2040-Pico, not Zero.

No, I intend RP2040-Zero, see image below:
[image]

> How do you connect it?

USB type-C from the zero to USB A to my PC.

> If you already have MicroPython installed on it then you should plug it without holding BOOTSEL

This solved the problem actually as I wrongly assumed somehow I had to hold BOOTSEL every time I connected it to program on it on Thonny, and not just for firmware updates 🙈. Thank you for the qwick response and sorry for my dum misinterpretation of BOOTSEL!

closed 2022-05-05 by Frank-99

aivarannamaa, 2022-05-06:
I'm glad I could help :)