thonny/thonny #2273: Thonny is giving error and not booting - Please Help

wrmspatel, 2022-05-06:
[image]
[image]

Thonny is giving error and not booting - Can you help?? Thank you.

aivarannamaa, 2022-05-06:
Thank you for the report!

I assume you have you been able to use Thonny on this computer before. Do you have an idea what may have changed? Different screen resolution perhaps?

You could try deleting Thonny's configuration file and start Thonny again with default options. The configration.ini should be under `~/.config/Thonny`

Whis OS and which Thonny version is this?

wrmspatel, 2022-05-11:
It is running on Raspberry Pi on a pi-top CEED.  Can't think of anything that might have changed. It was working fine on the CEED then started giving error.

wrmspatel, 2022-05-12:
Can anyone help with this?

aivarannamaa, 2022-05-13:
Did you try deleting the configuration file?

closed 2023-01-28 by aivarannamaa