thonny/thonny #2274: tempfile.py into the programm folder stops thonny to work with error messages .

Hermann12, 2022-05-07:
I struggle today with a strange failure, if I create a pythone file named tempfile.py thonny through error messages and stop directly to work. I searched long time, what's wrong with my program, but there isn't a failure. I can run it with spyder without error message, but thonny doesn't work. I think thony have a naming problem. also an empty "tempfile.py" file into the program directory stops thonny from work. I tried pre version 4 and version 3.
Better Error message will be helbful or maybe you can figure out the reason?

aivarannamaa, 2022-05-09:
Thank you for the report!

`tempfile` is a standard [library module](https://docs.python.org/3/library/tempfile.html), which Thonny's GUI is using. If you start Thonny inside the directory containing a module with the same nime, it tries to use this instead (that's how Python modules work).

Which Thonny version is this? Do you start Thonny via command line?

Could you send me the whole error message together with stacktrace?

Hermann12, 2022-05-09:
I would like create a tempfile.py module for my application: Gui_Phonebook: 
I created a tempfile.py file and got this Error message, if I run my GUI. even this tempfile.py isn't imported in any other python file. I thought first somthing with my code or import requests doesn't fit, but the same program without any changes work in Spyder without any trouble.
If I delete the file "tempfile.py" in the same directory all works without trouble.

Thonny 4.0.0b2 but I tried also the last stable version. Same issue.

%Run Gui_Phonebook.py
Traceback (most recent call last):
  File "D:\Daten\Programmieren\Fritz_04\Gui_Phonebook.py", line 43, in <module>
    from connectFritz import connectFritzBox
  File "D:\Daten\Programmieren\Fritz_04\connectFritz.py", line 3, in <module>
    import requests
  File "C:\Python39\lib\site-packages\requests\__init__.py", line 133, in <module>
    from . import utils
  File "C:\Python39\lib\site-packages\requests\utils.py", line 28, in <module>
    from ._internal_utils import to_native_string
  File "C:\Python39\lib\site-packages\requests\_internal_utils.py", line 11, in <module>
    from .compat import is_py2, builtin_str, str
  File "C:\Python39\lib\site-packages\requests\compat.py", line 62, in <module>
    from urllib.request import parse_http_list, getproxies, proxy_bypass, proxy_bypass_environment, getproxies_environment
  File "C:\Python39\lib\urllib\request.py", line 102, in <module>
    from urllib.error import URLError, HTTPError, ContentTooShortError
  File "C:\Python39\lib\urllib\error.py", line 14, in <module>
    import urllib.response
  File "C:\Python39\lib\urllib\response.py", line 14, in <module>
    class addbase(tempfile._TemporaryFileWrapper):
AttributeError: module 'tempfile' has no attribute '_TemporaryFileWrapper'

Hermann12, 2022-05-10:
I recognize that Thonny send also warnings about files in the program directory, which currently not referenced/ imported into the running program. Could this the reason for such a behavior?

aivarannamaa, 2022-05-13:
From the stacktrace it looks like your code depends on `requests`, which in turn depends on `urllib` in the standard library, which imports `tempfile` and assumes it gets the one in the standard library, not your tempfile.py.

Depending on how the code is run, the current directory or the directory of the main script is added as first item of `sys.path`.
If you don't see this problem with Spyder, then it may use different working directory or different approach for launching your program. You can find it out by adding following lines to the start of your program:

```python
import sys
import os
print("sys.path", sys.path)
print("Working dir:", os.getcwd())
```

closed 2023-01-28 by aivarannamaa