thonny/thonny #2278: thonny match_case: syntax error

Spiekes, 2022-05-09:
When I run:
# Building a simple match-case statement in Python
response_code = 400
match (response_code):
    case 200:
        print("Everything's peachy!")
    case 300:
        print("You're going somewhere else!")
    case 400:
        print("Uh oh. Are you lost?")
    case 500:
        print("Yikes. Something went wrong!")

I get:
Python 3.7.9 (bundled)
>>> %Run case.py
Traceback (most recent call last):
  File "C:\Users\hendr\case.py", line 6
    match (response_code):
                         ^
SyntaxError: invalid syntax

I don't get that error in: Idle (Python 3.10.2 (tags/v3.10.2:a58ebcc, Jan 17 2022, 13:59:55) [MSC v.1929 32 bit (Intel)] on win32)

paeryn, 2022-05-10:
Since match: was only introduced in Python 3.10 it shouldn't be a surprise that Python 3.7 sees it as a syntax error.

aivarannamaa, 2022-05-10:
That's right, the Python 3.7 coming with Thonny 3.3 does not support this constructor. 

If you need this, then you could try the latest beta of the upcoming 4.0, which comes with Python 3.10: https://github.com/thonny/thonny/releases

closed 2023-01-28 by aivarannamaa