thonny/thonny #2282: Thonny Error : Device is busy or does not respond.

eya1903, 2022-05-13:
I am  using an ESP32-CAM and Thonny's IDE runs the code properely.  But whenever I disconnect the chip and reconnect it or unplug the cable and return it this error shows every time : 

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - use Stop/Restart to interrupt more and enter REPL.
 
Even though I stop running the program and I disconnect the board before unplugging.

I have tried Ctrl+C, but nothing changed, I tried Ctrl+F2 but same error everytime. I also erased flash but nothing.

I tried to : Disconnect board > re-install the firmware > Click on reset button, this shows on terminal :

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

I really don't know if it's a board's problem or IDE.
Thank you for your time.

eya1903, 2022-05-16:
[SOLVED]

For anyone facing a similar problem, this is how i solved it ( I was really just testing and trying everything I can until i got a solution but really i don't know any explanation, sorry :) )

- Ctrl+D
- Connect GPIO0 to GND
- Reinstall firmware ( Run > Select Interpreter > Port or WebREPL : USB Serial Port (port). (COM7 for my case). ( Consider erasing flash before uploading).
- Install or update firmware
- Close window after uploading
- Disconnect GPIO0 from GND
- Click on Reset button 
- Stop/Restart Backend OR Ctrl+F2

I hope i added a little help for people who faced a problem like that and didn't find a solution anywhere.

aivarannamaa, 2022-05-17:
Thank you for the report!

Do I understand properly, that once you've used your device in Thonny and disconnect, you can't reconnect to it anymore? In this state, can you connect and get to the REPL using a generic serial program (eg. Putty, screen or tio)?

Do you have an endless loop in your main.py?

closed 2022-08-01 by aivarannamaa

Lunatik00, 2022-09-02:
I am having the exact same error, it worked once to reset the board but I have tried with 2 different boards and swapping the flash board, but nothing, it worked once and never again, but the board is available to do anything trough cli

aivarannamaa, 2022-09-03:
> but the board is available to do anything trough cli

Do you mean you can connect to it via a simple serial program (e.g. Putty or tio)? Which platform are you on? Do you close the connection from the other program before attempting to connect in Thonny?

oldbird9527, 2022-10-05:
Same error.  I am using ESP32-pico board.  this error occured when using UART function to connect serial port. 
The Only way to solve it is Reinstall firmware by Thonny.

Lunatik00, 2022-10-05:
>  Do you close the connection from the other program before attempting to connect in Thonny?

I used thonny to try to connect, it was available once and never again, but sometimes I can connect using things like picocom, but it is completely random if it can connect, obviously I only try one at the time. I use linux, to be completely honest, I think is the firmware where the error resides, at least for the esp32-cam, the same board works fine using the arduino interface, so it is not the board, and I have used other boards with thonny in the past, so I don't really think it is a Thonny problem, maybe it would be a good addition to Thonny to have vetted firmwares for boards that are compatible with micropython, that way we could discard the case where it is a firmware problem. One of the most common firmwares used for ESP32-cam is [this one](https://github.com/lemariva/micropython-camera-driver). I didn't answer before because I was out for a while and I can't do testing in the board for now.

jouellnyc, 2022-11-20:
Hi. Google brought me here. Similar issue for me.

 Just got a brand new d1 mini. gtkterm connect but thonny does not after a fresh install and hitting reset button/etc.

Running esp8266-512k-20221118-unstable-v1.19.1-705-gac5934c96.bin

Hope this helps.

[image]

ExcaliburAvalon, 2022-11-25:
same error！When I use arduino, I can program and open ports, and when I use thonny, I can't save the program to esp32, and it replies to me with this message
Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython / firmware;
  - make sure the device is not in bootloader mode.

I tried the following method, but nothing happened.
Click on Reset button
Stop/Restart Backend OR Ctrl+F2

aflores01, 2023-01-06:
i'm facing this issue with an ESP32 38pin and the only way I fix it was to flash it with mu editor again istead of cli

magixyu, 2023-03-02:
I have the same issue with ESP32-CAM. I did a fresh install and I can only access it via the "Serial port monitor" in Arduino IDE. Thonny, Mu and Putty all doesn't work. 

Any ideas?

magixyu, 2023-03-02:
b.t.w. I tried to install it using Mu Editor, cli and Thonny, the result is the same.

RoanArraes, 2023-03-25:
I got the same problem and I recovered my esp8266 deploying the micropython firmware again.  Followed this steps:
https://docs.micropython.org/en/latest/esp8266/tutorial/intro.html

selio0299, 2023-04-09:
Tuve el mismo problema con la placa esp32, mi solución fue tener un botón (tercer botón) a un costado de la placa ESP32. Al presionar este boton, Thonny lo reconoce como a su hijo ;)

sscro, 2023-05-01:
> same here, 8266-D1 mini, micropython v1.20.0, Thonny 4.0.2 when introducing a time.sleep (or utime.leep, whichever...) in a while True loop there is no way to stop the program neither by "stop" button nor by ctrl+c. Shell reports same error/comment as in posts above. Tried different combination of Thonny / micropython up to v 3.1 for the former and 1.7 for the latter Any advise?

sorry to alarm you guys:
a deeper debug found that the issue was due to a bad (very bad) use of try-except error catch!
After setting it, everything now properly works!
sorry again for that ;)

tuncaycaner, 2023-05-03:
If anyone having this issue, consider switching usb port. Mine was fixed that way.

victorkemp, 2023-05-23:
Same problem. I solved it by using a new Pi Pico *and* re-installing Thonny with the same version (4.0.2).

covalspace, 2023-09-01:
I had the same problem. The issue is because the device cannot use REPL while in bootloader mode. 

These are the steps to fix it assuming you have already flashed the device with MicroPython firmware:

Press RESET/RESTART on the board (this step takes the device out of bootloader mode and assigns a new port)
Go to Run -> Configure Interpreter 
Select the new port 
Click OK

If you don't select the new corresponding with the Espriff device it will say the device can't be found because the port tends to be relative to bootloader mode. 

Hope this helps!

sany3001, 2023-11-08:
I have the same problem with RPi Pico W. No problem on Windows. No problem on Linux Mint yesterday either. Today, I am facing problem described above (in Linux only).
`
Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - use Stop/Restart to interrupt more and enter REPL.`

I have:

- changed USB port
- reprogrammed MicroPython to RPI_PICO_W-20231005-v1.21.0.uf2

sany3001, 2023-11-08:
Ok "Send EOF / Soft reboot" in menu "Run" did help. :-)

mister0wl, 2023-11-12:
I had the same problem with an A9G board with MicroPython v1.9.4, Windows 10, Thonny 4.1.2. Nothing helped until 
1. I flashed the board to the AT-mode firmware, then 
2. back to the micropython firmware, then 
3. I had to remove Thonny, then 
4. remove the directory C:\users\<name>\AppData\Roaming\Thonny manually, then 
5. install Thonny again. 

If any of these steps skipped, it felt back to the error message mentioned in the first post.

fbarralaga, 2023-12-29:
I had the same problem with Thonny trying to connect with a ESP8266 and found a solution using older version of the firmware (I used one from 2020) because the ESP8266 module I was working with, was old.

PsychLH2022, 2024-03-02:
I faced the same issue. I changed a usb cable and then it works again.

Gasebelo, 2024-03-23:
SOLUTION

Just install the most recent Firmware again on Thonny and when you start to download it, keep the Reset button pressed until it finish all the process, and then you will be able to use it again.

THIS HELPED ME TO ERASE AN ARDUINO CODE!!!

manish33scss, 2024-04-01:
i think i found it, so press CTRL+F2 and as soon as it breaks start pressing CTRL+C, you should be blazing fassssst !!!

toddmo, 2024-04-17:
> Tuve el mismo problema con la placa esp32 (ttgo t-display), mi solución fue tener un botón (tercer botón que al parecer es el Reset) a un costado de la placa ESP32. Al presionar este boton, Thonny lo reconoce como a su hijo ;)

I had the same problem with the esp32 board (ttgo t-display), my solution was to have a button (third button which is apparently the Reset) on the side of the ESP32 board. By pressing this button, Thonny recognizes him as his son ;)

Frog-man-sec, 2024-04-21:
I faced the same issue. I solved it with **USB-TTL converter.** 
[image]
[image]

mrPrimezx, 2024-05-05:
[image]
ESP WROOM 32 devkit V1 board show above issue with thonny ide, but board work fine with arduino ide.How to use this board with thonny ide micropython?

mrPrimezx, 2024-05-05:
[image]
Received different error message after connect 3v3 pin with EN for second!

gogutier, 2024-06-05:
Can confirm that I solved the issue by connecting the power to 3v instead of 5v. The classic programming board of the esp32CAM was the problem I think.

VickyZhr, 2024-07-01:
> Same problem. I solved it by using a new Pi Pico _and_ re-installing Thonny with the same version (4.0.2).

same.

stefaniework, 2024-07-12:
> 我想我找到了它，因此按下 CTRL+F2，并且在它中断后立即按 CTRL+C，你应该会非常快！！！

厉害了，确实可以，我在boot中写了死循环也有sleep代码，不知道哪块引起的问题导致的无法连接

Kaizonium, 2024-07-22:
> [SOLVED]
>
> For anyone facing a similar problem, this is how i solved it ( I was really just testing and trying everything I can until i got a solution but really i don't know any explanation, sorry [...]

Thank you so much. I followed your instructions carefully and it works

kf4mot, 2024-07-25:
> Thank you so much. I followed your instructions carefully and it works

Also worked for my esp32. I had 2 of the same boards, but one that didn't work had some extra wiring that included a pullup on gpio0.

kresimirhecimovic, 2024-11-29:
I solved the problem by NOT using builtin LED. 
My ESP32 (made by AZ-Delivery) has both TX pin and builtin LED pin set to GPIO01.
I am using external LED instead of builtin LED and everything works fine.