thonny/thonny #2283: Connecting Rpi4 and Rpi pico via uart serial

LoryCraft1, 2022-05-16:
Hi everyone
i'm new to this, and i want to have access to a shell on mi 4 via rpi pico, i have this error:

- Windows 10
- Thonny 3.2.7

PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 689, in _cmd_Run
    self._execute(cmd["source"])
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 441, in _execute
    return self._process_until_raw_prompt(capture_output)
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 532, in _process_until_raw_prompt
    self._check_for_side_commands()
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 635, in _check_for_side_commands
    self._submit_input(cmd.data)
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')
Traceback (most recent call last):
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 689, in _cmd_Run
    self._execute(cmd["source"])
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 441, in _execute
    return self._process_until_raw_prompt(capture_output)
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 532, in _process_until_raw_prompt
    self._check_for_side_commands()
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 635, in _check_for_side_commands
    self._submit_input(cmd.data)
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\zivob\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')

aivarannamaa, 2022-05-17:
Thank you for the report!

Thonny 3.2.7 is rather old. I recommend trying https://github.com/thonny/thonny/releases/tag/v4.0.0b3

Also, Pico's MicroPython 1.17 and 1.18 have some nasty problems which become apparent, when using Thonny. I recommend upgrading to latest nightly build: https://micropython.org/download/rp2-pico/

When the problem persists, then make sure you're not trying to use the UART which is tied to MicroPython REPL (https://forum.micropython.org/viewtopic.php?t=6182)

LoryCraft1, 2022-05-21:
ok, i'l let you know
Thanks

closed 2023-01-28 by aivarannamaa