thonny/thonny #2288: Thonny struggles to open the last modify file when raspberry pi reboot

viafx24, 2022-05-18:
Dear Thonny Teams,

Thanks for this great software. I have a little issue using Thonny 3.3.10 under raspbian 10 buster (using VNC to connect to it). When i work on a file and reboot the raspberry pi, it do not always remember the last file i was working on. And thus i need to reopen it. Sometimes, it succeed to remember the last file but most of the time, it doesn't not and open an much older one at startup. This is not a big issue but it would be very convenient to be fixed. All the best. Via_Fx_24.

aivarannamaa, 2022-05-18:
Thank you for the report!

When you reboot, do you close Thonny first? If not then it may be killed by shutdown procedure and not get the chance to save the open files as it would during normal close.

viafx24, 2022-05-18:
Thanks a lot for your reactivity and for your answer that perfectly solved my problem, aivarannamaa ! This "problem" bothers me since more than one year and i would never have imagined that the solution was that simple! I dont know why i shutdown the raspi without stopping thonny correcly first. I didn't even mind it! Thanks a lot and all the best.

closed 2022-05-19 by aivarannamaa