thonny/thonny #2292: Copy, paste sometimes broken.

viafx24, 2022-05-19:
Dear Aivar Annamaa,

I wrote you yesterday and you solved my problem in a minute: Thanks again!  Today, I would like to report two other issues I also encounter sometimes with Thonny (which is the best!). I'm using  Thonny 3.3.10 under raspbian 10 buster (using real VNC to connect to it from windows). Sometimes, the "copy and paste" doesn't work. For instance when I want to copy all code (or a single line of code) of a python script under Thonny to paste it in Onenote under windows, sometimes it works, sometimes it doesn't. I thought it was a problem of Linux/raspbian and Windows compatibility issue or a realVNC issue but when i copy text from the terminal or from Geany (another IDE) , i always can paste in windows. Thus, i concluded that it may be an issue with Thonny. This problem looks to appear randomly, thus it may be not easy to solve: sometimes I "copy and paste" from Thonny without problem. 10 second latter, it failed. If I close and reboot Thonny, sometimes, it works again, sometimes not. I have not idea where come from the problem.  Maybe related to other issue: Copy, paste broken. #2171. Maybe not.  I have written the second issue (related to breakpoint) in a second post. All the best. Via_Fx_24.

aivarannamaa, 2022-05-20:
Thank you for the report!

The copy-paste problem is tough one, because I haven't been able to reproduce it myself. I'm leaving it for now, with the hope that Tk 8.6.11 in Bullseye works better than 8.6.9 in Buster in this respect.

viafx24, 2022-05-20:
Yeah, I was guessing that it was a tough one.  I found a trick not particularly beautiful but working. I copy my python code from Thonny and I paste it in Geany (this "copy and paste" is working). Then I copy again from Geany and paste it in windows (through realVNC) and that is working again. I will wait new version of Tk and Bullseye to see if this bug disappears. Thanks for you reply and all the best. Via_Fx_24.

aivarannamaa, 2022-05-20:
> I will wait new version of Tk and Bullseye to see if this bug disappears

You probably know it, but the latest version RPiOS is already based on Bullseye.

viafx24, 2022-05-20:
Thanks, no i didn't know. I will probably not reinstall raspbian right now to check the new version. But  when i will update Raspbian, i will test if the "copy and paste through VNC" is working again in Thonny and give feedback. Thanks. Via_Fx_24.

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem with Thonny 4.0

closed 2023-01-28 by aivarannamaa

viafx24, 2023-01-29:
Thanks for your work. I don’t currently working with python (and thus thonny) but as soon as I will go back to programming, I will install thonny 4 and try again the copy and paste to let you know.

 

All the best,

 

Guillaume

 

Cet e-mail a été vérifié par le logiciel antivirus d'Avast.
www.avast.com