thonny/thonny #2293: Hard to locate the area where to click to create a new breakpoint and hard to remove it as well

viafx24, 2022-05-19:
Dear Aivar Annamaa,

This is the second issue I encounter with Thonny.  I'm using Thonny 3.3.10 under raspbian 10 buster (using real VNC to connect to it). When i want to debug my code, i like to use breakpoints. But it always take me 10 or 15 clicks to locate the exact area that will trigger the arrival of the breakpoint (a red dot appears). I regularly use other IDEs to set breakpoint and i don't have this problem. It maybe related to the mouse cursor of realVNC or maybe not. I have to test several locations: on the line number, on the left of the line number, a bit more on top, a bit more bottom,  a bit more on the right until the breakpoint appears. Same happen when i want to remove the breakpoint: I cannot just click on the red dot: I need to try 10-15 locations before it disappears. I'm not totally convinced that it is a VNC cursor problem since for all other stuff (under Thonny or other GUI), i dont have any issue. Thanks for Thonny which is great and all the best.  Via_fx_24.

aivarannamaa, 2022-05-20:
I assume you assumed it takes a single click to set a breakpoint, and, to be fair, it's a fair assumption.

When I designed the debugger, I thought it's better to require a double-click, so that there's smaller chance that a beginner sets a breakpoint accidentally and needs to wonder what happened and if he broke the computer. I'm not sure anymore. I may change this before the final release of Thonny 4.0.

viafx24, 2022-05-20:
You are a life saver! Yeah, it was that: with double click, it perfectly works! Yes, it may be a good idea to consider changing this behaviour since most IDE use single click (I use codeblock, Visual studio, VS code, Matlab among others) and for folks used to IDEs with "one click", it may be hard to figure out that setting a breakpoint need two clicks. Even if your argument to protect beginners from mistakes is good too. Thanks again and all the best. Via_Fx_24.

aivarannamaa, 2022-08-13:
In the next beta, Thonny accepts single click for setting or removing a breakpoint.

closed 2022-08-13 by aivarannamaa

viafx24, 2022-08-13:
Thanks to let me know. 

 

All the best,

 

Guillaume

 

Cet e-mail a été vérifié par le logiciel antivirus d'Avast.
www.avast.com