thonny/thonny #2294: Suggestion for a new feature for Thonny: smart indent ?

viafx24, 2022-05-19:
Dear Aivar Annamaa,

A third post tonight about Thonny. Here it is not an issue but a suggestion for a new feature: smart indent. Maybe it is not the place to post suggestions of new functionnality, i'm not very familiar with github, sorry if I should have post this suggestion in another section.
I use "smart indent" with shortkey under visual studio code (shift alt F) or under matlab (ctrl A followed by ctrl +i) and this allows (in case of you dont already know) to automatically indent all the code (maybe it's not easy in python since this latter uses indentation in place of bracket). I would thus appreciate this feature in future release of Thonny. That'is all for tonight! All the best. Via_Fx_24.

aivarannamaa, 2022-05-20:
In a sense smart indent is already implemented -- for example when you type `if blah:` and press ENTER, the next line will be indented.

I don't see how automatic _indent_ would be possible at other places with Python, because indentation has meaning and therefore should be writer's decision.

In case you meant more general automatic _format_, which would reorder code nicely without affecting the meaning, then there's a plug-in which uses popular `black` formatter: https://pypi.org/project/thonny-black-format/

viafx24, 2022-05-20:
Thanks. I installed thonny-black-format  and found that it was struggling solving small issues of indent (a space in excess for instance) but correclty removing blank lines in excess (basic code formatting). Thus it was not the exactly desired smart indent that i'm used too in C++ under VS code or under Matlab. However, i have tried to see if Visual studio code in Python does a better job and it looks like it does not: it also can't solving small typos of space in excess but succeed removing blank line in excess. As you said, i guess that it's harder in python since indent has meaning. Never mind! I will continue my life without smart indent in python! Thanks again for your answer and all the best. Via_Fx_24.

closed 2022-05-20 by aivarannamaa