thonny/thonny #2296: Avoid location-dependent hard-coded absolute path on Mac

ypiguet-epfl, 2022-05-20:
On Mac, Thonny is launched by a script whose shebang is the absolute path of its own Python interpreter. This is typically
```
#!/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python
```
but depends on the location of Thonny itself. This must explain why an installer is required: the [`postinstall` script](https://github.com/thonny/thonny/blob/master/packaging/mac/scripts/postinstall) sets the shebang. Then Thonny becomes broken when it's moved. When possible, Mac applications are installed just by dragging their icon in the Finder.

Suggestion: replace the Python launch script which requires the absolute path of the Python interpreter with a plain /bin/sh script which finds dynamically the location of Python and starts it with the Python script. For example a file named "thonny" containing
```
#!/bin/sh

script_dir=`echo "$0" | sed -E 's/\/[^\/]*$//'`

python_path="${script_dir}/../Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python"
thonny_py_path="${script_dir}/thonny.py"

"$python_path" "$thonny_py_path"
```

`thonny.py` would contain the current content of `thonny` without the shebang (neither required nor used).

aivarannamaa, 2022-05-21:
Thank you for the suggestion!

Have you tried this yourself?

I have struggled quite a bit for getting Thonny's application bundle working properly eg. displaying "Thonny" as application name at the menu bar instead of "Python". With some approaches it couldn't launch PyGame applications properly, other approaches used to have other problems. Before Catalina I used to distribute Thonny as drag-n-drop dmg, but Catalina brought problems which I couldn't solve other than introducing the installer and using the same approach for launching as Python's IDLE.

ypiguet-epfl, 2022-05-21:
@aivarannamaa I've tried by modifying the installed Thonny on Monterey. Then you can move it elsewhere, it can still be launched and you can import its modules. I haven't tested much more than that.

I'd expect modifying the application at install time or later will be more and more restricted. Is compiling all the modules to pyc files worth it? or required?

aivarannamaa, 2022-05-21:
The compilation to pyc was not the main reason for introducing pgk installer.

With Catalina the main problem was OS not trusting the app even though it was notarized. Introducing installer was one step in resolving this issue. Therefore I can't be sure your experiment proves the same approach works after notarized dmg is downloaded from the internet.

Unfortunately my Mac is stuck with Catalina, so I can't easily test a approach on later macOS-es. Therefore I'll continue with pkg installer for Thonny 4.0 and I'll postpone this issue at least until Thonny 4.1.