thonny/thonny #2297: Pygame showing 'Backend terminated or disconnected. Use 'Stop/Restart' to restart.'

Flufiku, 2022-05-21:
I'm using Thonny to make (pretty shitty) pygame scripts, and every time I use the pygame.quit() function to end the script, it shows 'Backend terminated or disconnected. Use 'Stop/Restart' to restart.'. It only happens in pygame (other scripts work fine) and it only happens after using the quit function. I'm using thonny since Autumn 2021 and this started happening ca. 2 weeks ago.
(Os: Windows 10; Python version 3.7.9)

Thanks in advance and sorry for any spelling errors.

aivarannamaa, 2022-05-22:
Thank you for the report!

Did you upgrade Thonny or Pygame 2 weeks ago? Did the `quit`-function work without this message before?

Flufiku, 2022-05-23:
No, I didn't update it.

Yes, it worked before

aivarannamaa, 2022-05-25:
Which Thonny version and which Pygame version are you using?

closed 2023-01-28 by aivarannamaa