thonny/thonny #2299: Virus Warning for numpy

jochen-rath, 2022-05-24:
Hello

I am running thonny behind a proxy without admin rights. I tried to install numpy in the system shell with this command:
python -m pip  --proxy=http://proxyserver:prot/ install numpy

the installation was successful, but afterwards I found this message from my system
Win32:Evo-gen [Susp]: /home/username/Windows/Remote.V2/AppData/Python/Python37/site-packages/numpy/linalg/_umath_linalg.cp37-win32.pyd

is it possible, that there is in Virus numpy?

Kind regards
Jochen

aivarannamaa, 2022-05-24:
> is it possible, that there is in Virus numpy?

I don't find it likely, but to be sure, you should ask this in Numpy forum or mailing list. If you do, then I recommend adding more details (antivirus name and version, detailed report provided by antivirus).

jochen-rath, 2022-05-25:
Thanks for the replay.
Do I understand correct, that Thonny uses the official numpy sources and do not modify numpy on its on, to fit it into its environment?

The problem is, that I am a teacher in germany in an school environment called iServ. I am trying to determine, who I need to contact with the problem. I already tried to contact our admin, to get more information, but didn't get any reply yet. 
I would like to give proof, that it is a problem with iServ and not numpy.

aivarannamaa, 2022-05-25:
> Do I understand correct, that Thonny uses the official numpy sources and do not modify numpy on its on, to fit it into its environment?

That's right. Actually Thonny doesn't even know about Numpy. When you install numpy in Thonny's package manager, then it simply executes `pip install numpy`, which will download suitable distribution file for PyPI and installs it. On popular platforms it will download a pre-built binary whl. 

It looks like the latest version of Numpy doesn't come with binary wheels for Python 3.7 (https://pypi.org/project/numpy/#files), so if you have Python 3.7 (eg. bundled with Thonny 3.3) and pip fails compiling the sources, you may want to use version 1.21 (https://pypi.org/project/numpy/1.21.6/#files).

Thonny 4.0 (currently in beta) is bundled with Python 3.10 and there are binary wheels available for it.

closed 2023-01-28 by aivarannamaa