thonny/thonny #2300: Package manager

88cy, 2022-05-26:
Installed the package simple calculator on Thonny IDE. However, when I adhered to the written codes to run simple calculator, I only get a runtime error saying that simple calculator was not found, or that it does not recognize the word from calculator.simple import SimpleCalculator because that is written in the example or source code for running Simple Calculator.

aivarannamaa, 2022-05-26:
Do you mean this package: https://pypi.org/project/simplecalculator/ ?

Are you sure you installed it via Thonny's package manager and not via Plug-ins manager?

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem and can answer my question.

closed 2023-01-28 by aivarannamaa