thonny/thonny #2301: Bad Spanish traduction in package manager

javacasm, 2022-05-26:
When an installed packaged can be upgraded, the button for upgrade it show "Actualizado" ("Upgraded") instead of "Actualizar" ("Update")

Change propposed 

File thonny/locale/es_ES/LC_MESSAGES/thonny.po

line 371
#: thonny/plugins/pip_gui.py:76
msgid "Upgrade"
<<msgstr "Actualizar"
>>msgstr "Actualizado"

aivarannamaa, 2022-05-26:
Thanks for noticing this!

You can fix this yourself after joining Thonny project in POEditor: https://poeditor.com/join/project/Gh188fdYH6

Before next release I'll update translation files from POEditor's current state.

javacasm, 2022-05-29:
Done!! I made the change in poeditor
Thanks

aivarannamaa, 2022-05-31:
Great, I'll use updated translation files before the next release.

closed 2022-08-14 by aivarannamaa